odriverobotics/ODrive #589: v0.5.2 Version Strings Are Wrong

CthulhuLabs, 2021-06-26:
Something is wrong wiht the versioning info in the v0.5.2 release. I have updated with "odrivetool dfu" as well as with openocd with a st-link. Attached is the output from doing so. 
[Odrive Upgrade Output.txt](https://github.com/odriverobotics/ODrive/files/6720245/Odrive.Upgrade.Output.txt)
It looks like the v0.5.2 release has the version string set to v0.5.1-dev.

samuelsadok, 2021-06-28:
Yes the firmware binary contains the wrong version number, this will be fixed in 0.5.3.

CthulhuLabs, 2021-06-28:
You should mention that in the discourse release thread:

https://discourse.odriverobotics.com/t/firmware-0-5-2-release-thread/7151

I have seen a number of people confused by this on Discord.

Wetmelon, 2021-07-18:
Done.  Thanks

closed 2021-07-18 by Wetmelon