odriverobotics/ODrive #591: Hangs when connecting to multiple odrives using find_any()

Patrick-Thomas, 2021-06-29:
I'm trying to control two ODrives inside a Python3 script. Both ODrives (v3.6-56V) are running the latest firmware version (fw-v0.5.2), and both appear when I open odrivetool (v0.5.2.post0). In Python I can instantiate either ODrive using find_any(serial_number="xxx"), however when I try to instantiate them sequentially I get a couple of warnings ([LEGACY_PROTOCOL] received unexpected ACK: 129) and the script hangs.

I am running Ubuntu 18.04, and the ODrives are not connected via a USB hub. I've tried each of the usb_cdc_protocol settings for each ODrive but none of them work.

Fallstop, 2021-07-01:
This also affects me, on Ubuntu 21.04, fw-v0.5.2, ODrives v3.5-24V.
Example code to trigger the bug:
```
odrv_0 = odrive.find_any(serial_number="XXX")
odrv_1 = odrive.find_any(serial_number="XXX")
```

This appears to be a regression in Odrivetool v0.5.2.dev0, as this version does connect to both odrives.

samuelsadok, 2021-07-02:
This is because `find_any()` starts a `fibre.Domain` internally which remains open for the lifetime of the ODrive objects. Since Fibre currently doesn't support multiple concurrent domains, the second `find_any()` will run into problems.

I rewrote `find_any()` now so that only one single Domain is started that remains open until the program is terminated. All subsequent `find_any()` calls make use of this first Domain.

I pushed the fix to `fw-0.5.3`.

closed 2021-07-02 by samuelsadok