odriverobotics/ODrive #593: `odrivetool generate-code` is broken in v0.5.2

Jascha-N, 2021-07-08:
**Describe the bug**

Running `odrivetool generate-code ...` fails with the following output:
```
>odrivetool generate-code -t endpoints.inc.j2 -o endpoints.inc
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
D:\Applications\anaconda3\Scripts\odrivetool in <module>
    191         my_odrive = odrive.find_any(path=args.path, serial_number=args.serial_number,
    192             channel_termination_token=app_shutdown_token)
--> 193         generate_code(my_odrive, args.template, args.output)
    194
    195     elif args.command == 'backup-config':

D:\Applications\anaconda3\lib\site-packages\odrive\code_generator.py in generate_code(odrv, template_file, output_file)
     32
     33 def generate_code(odrv, template_file, output_file):
---> 34     json_data = odrv._json_data
     35     json_crc = odrv._json_crc
     36

AttributeError: 'anonymous_interface_2016658132128' object has no attribute '_json_data'
```

**To Reproduce**
Run `odrivetool generate-code` on any inputs.

**Expected behavior**
It should work? Or be removed if it is superseded. It used to work fine on v0.5.1.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - odrivetool version: v0.5.2.post0

samuelsadok, 2021-07-09:
Thanks for the heads up.

This feature is no longer supported so I removed it on the fw-v0.5.3 branch.

You can now instead use [odrive-interface.yaml](https://github.com/odriverobotics/ODrive/blob/master/Firmware/odrive-interface.yaml) as data source for code generation, so you don't need to have an ODrive connected. [interface_generator.py](https://github.com/odriverobotics/ODrive/blob/master/tools/fibre-tools/interface_generator.py) parses this file and normalizes it into a form where it can be given to a Jinja2 template. [Here's](https://github.com/odriverobotics/ODrive/blob/master/Firmware/fibre-cpp/interfaces_template.j2) an example template file. You can find more examples by looking for the *.j2 file extension in this repo.

closed 2021-07-09 by samuelsadok

Jascha-N, 2021-07-11:
Ok, thanks. But the docstring for `generate_endpoint_table` says it will be deprecated in the future. How am I supposed to generate endpoint bindings for my embedded C project in a future-proof way?

samuelsadok, 2021-07-12:
The deprecation warning is because some day the endpoint-based protocol will be replaced by a more powerful low level protocol. However this has not yet been rolled out.

In the future the recommended way to interface with the ODrive will be through [fibre-cpp](https://github.com/samuelsadok/fibre-cpp/tree/devel) (or its language bindings). However this still needs some time to converge towards a stable API. Currently the client side of fibre-cpp is intended for language bindings and not optimized for direct use in a C/C++ application. It also uses heap allocation. To fix that, we'll need to autogenerate client side mock classes from the interface yaml.

In the meantime using `generate_endpoint_table` is probably the most practical choice for the medium term.

Jascha-N, 2021-07-12:
So do I run `python interface_generator.py ... --generate-endpoints ODrive3` to generate all endpoints? And do the generated endpoint IDs correspond to the actual native endpoint IDs? Because the IDs generated change based on the interface you choose to generate endpoints for. For example, if I do `--generate-endpoints ODrive`, the ID for `clear_errors` will be 89. If I do `--generate-endpoints ODrive3`, the ID for `clear_errors` will be 758. What do these IDs stand for, if they are not consistent?

Also, how do I obtain the equivalent of `json_crc` without using Wireshark...? This constant used to be available in the old code generator.

samuelsadok, 2021-07-14:
Yes `--generate-endpoints ODrive3` is correct. You can consider `ODrive` a base class for both `ODrive3` and `ODrive4` (once available). These ODrives have different hardware and thus a different object tree which is why the endpoint IDs differ.

The endpoint IDs can change between firmware and hardware versions but not between ODrives of the same firmware and hardware version.

The firmware uses `calc_crc16()` (crc.hpp) to calculate `json_crc`. It should be easy to port from C++ to C. Otherwise you can use the old Python function from [here](https://github.com/odriverobotics/ODrive/blob/fw-v0.5.1/Firmware/fibre/python/fibre/protocol.py#L56).

Jascha-N, 2021-07-15:
So I have tried to create a Python script to calculate the JSON CRC during the build process of my project. For this I used the simple template `[[embedded_endpoint_definitions | tojson]]`. First off, there is whitespace in the generated JSON so that doesn't work, because it will result in the wrong CRC. So I decode the JSON in Python and then encode it again without whitespace, before calculating the CRC. However this is not the only issue. The order of the keys in the dictionaries is lost in the conversion with `tojson` and I would have to rebuild every dictionary recursively in the correct order and encode the whole bunch. Is there really no easier way? Could we not just expose `json_crc` to the templates like it used to be? Or at least a template filter that returns the correctly ordered and formatted JSON? The only somewhat easy solution I can think of is compile and run a tiny C program during the build process that uses `[[embedded_endpoint_definitions  | to_c_string]]`, but that is still overly complicated. I really do not want to calculate the static JSON CRC at runtime...

samuelsadok, 2021-07-16:
You could split the function `to_c_string` into two parts: `json_to_string` and `c_string_escape` and then you could use `json_to_string` in combination with the CRC function I referenced earlier to expose `json_crc` to the template. Or leave `to_c_string` as it is and just use it as an orientation to get the correct JSON string.