odriverobotics/ODrive #596: ASCII response does not end with new line when checksums are used

Jascha-N, 2021-07-19:
**Describe the bug**
In UART ASCII mode, when using checksums (`command *42`), the ODrive will not end the response with a new line. This makes it hard to find the end of a response.

The relevant line:
https://github.com/odriverobotics/ODrive/blob/19e9b39b88ef13c4179fb3279239b0cb163cbfe6/Firmware/communication/ascii_protocol.cpp#L54

**To Reproduce**
Send any command with a correct checksum that causes a response. Example:
```
> r vbus_voltage *93\n
< 48.065918*33
```

samuelsadok, 2021-07-22:
Thanks for the report. I fixed this on the `fw-0.5.3` branch, can you check if that solves the issue?

Wetmelon, 2021-09-13:
Closing as fw 0.5.3 has released.  Please reopen if it's still a problem.

closed 2021-09-13 by Wetmelon