odriverobotics/ODrive #599: Motor goes to 0 when starting closed loop control with trap traj enabled

AzazKamaz, 2021-08-11:
**Describe the bug**
When starting closed loop control with trap traj enabled motor immediately goes to zero pos

**To Reproduce**
1. Restart ODrive (maybe optional)
2. Turn motor to get `pos_estimate != 0`
3. Set `input mode` to `trap traj` and `control mode` to `position control`
4. Set requested state to closed loop control
5. Motor goes to zero pos

**Expected behavior**
Motor should stay on current pos. Input pos is right but motor somewhy goes to zero.

**Desktop (please complete the following information):**
 - OS: MacOS but doesn't matter
 - odrivetool v0.5.2.post0
 - ODrive v3.6 56v with firmware fw-v0.5.2

**Additional context**
Looks like there is some bug in setting `input_pos_` and `pos_setpoint_` with starting trap traj. So problem is here when using CAN bus too. Now I am trying to fix it but code looks right for me.

AzazKamaz, 2021-08-12:
For now my code looks like (communicates with ODrive using can bus)
```cpp
void ODrive::enable(bool trap_traj) {
  set_modes(CONTROL_MODE_POSITION_CONTROL, INPUT_MODE_PASSTHROUGH);
  delay(50); // set modes can frame have no callback so wait 50ms

  // send required state can frame and wait for heartbeat with current state = required
  set_state(AXIS_STATE_CLOSED_LOOP_CONTROL, true);

  if (trap_traj) {
    delay(50); // Wait 50ms to be sure that control loop has started
    set_modes(CONTROL_MODE_POSITION_CONTROL, INPUT_MODE_TRAP_TRAJ);
    delay(25); // again wait for sure because there is no callback
  }
}
```
And even with such logic it sometimes (rarely) goes to zero on enable. Also functions used here are communicating with 6 axes on 3 ODrives

Wetmelon, 2021-08-15:
Ok so to confirm,

It starts at some non-zero `pos_estimate`, then it suddenly goes to `pos_estimate = 0` although `input_pos != 0`?

AzazKamaz, 2021-08-16:
Yes. So the reason is:
```cpp
bool Axis::start_closed_loop_control() {
    ...
    CRITICAL_SECTION() {
        ...
        // To avoid any transient on startup, we intialize the setpoint to be the current position
        // note - input_pos_ is not set here. It is set to 0 earlier in this method and velocity control is used.
        if (controller_.config_.control_mode >= Controller::CONTROL_MODE_POSITION_CONTROL) {
            std::optional<float> pos_init = (controller_.config_.circular_setpoints ?
                                    controller_.pos_estimate_circular_src_ :
                                    controller_.pos_estimate_linear_src_).any();
            if (!pos_init.has_value()) {
                return false;
            } else {
                controller_.pos_setpoint_ = *pos_init;
                controller_.input_pos_ = *pos_init;
                float range = controller_.config_.circular_setpoint_range;
                steps_ = (int64_t)(fmodf_pos(*pos_init, range) / range * controller_.config_.steps_per_circular_range);
            }
        }
        controller_.input_pos_updated();
        ...
    }

    // In sensorless mode the motor is already armed.
    if (!motor_.is_armed_) {
        wait_for_control_iteration();
        motor_.arm(&motor_.current_control_);
    }

    return true;
}

bool Motor::arm(PhaseControlLaw<3>* control_law) {
    ...
    CRITICAL_SECTION() {
        ...
        // Reset controller states, integrators, setpoints, etc.
        axis_->controller_.reset();
        ...
    }

    return true;
}

void Controller::reset() {
    pos_setpoint_ = 0.0f;
    ...
}
```

So execution goes like:
1. Set `input_pos_` and `pos_setpoint_` to current position
2. Wait for control loop end (trajectory is planned from `pos_setpoint_` to `input_pos_` and then set `trajectory_done_ = true` so `pos_setpoint_` will not be updated anymore
3. Reset controller so `pos_setpoint_ = 0`
4. Motor moves to current `pos_setpoint_`

Suggested patch is to move `input_pos_` and `pos_setpoint_` setting from `Axis::start_closed_loop_control()` to `Controller::reset()`. I will make a pull request a bit later

AzazKamaz, 2021-08-16:
Also I don't sure that it is correct to do this on planning trajectory:
```cpp
void Controller::move_to_pos(float goal_point) {
    axis_->trap_traj_.planTrapezoidal(goal_point, pos_setpoint_, vel_setpoint_,
                                 axis_->trap_traj_.config_.vel_limit,
                                 axis_->trap_traj_.config_.accel_limit,
                                 axis_->trap_traj_.config_.decel_limit);
    axis_->trap_traj_.t_ = 0.0f;
    trajectory_done_ = false;
}
```

`pos_setpoint_` and `vel_setpoint_` are expected motor position and velocity at moment. Like if I have `INPUT_MODE_PASSTHROUGH` at moment and I am moving 100 turns clockwise and then start `INPUT_MODE_TRAP_TRAJ ` it will plan trajectory starting in current `pos_setpoint_ = input_pos_` that can be 90 turns far away from current position

samuelsadok, 2021-09-01:
Thanks for investigating this.
Since `pos_setpoint_` is already set in `start_closed_loop_control()` we don't need to reset it in `Motor::arm()`. So I removed the line in 99713725ace0b0ca25e78fb80b3ed017b3811e28 and it fixes the snap-to-zero issue you're describing.

closed 2021-09-03 by samuelsadok