odriverobotics/ODrive #602: "Oh no odrv0 disappeared" message on Ubuntu 20.04 after odrv0.save_configuration()

mirano-Pat, 2021-09-03:
After "odrv0.save_configuration()", I get the message "Oh no odrv0 disappeared" message and after few seconds I get a message "Reconnected to ODrive 2xxxxxxxxxx8 as odrv0."

ODrive board: 24V ODRIVE V3.6  
Frimware: v0.5.2  
OS: Ubuntu 20.04 LTS

```bash
$ odrivetool

Connected to ODrive 2xxxxxxxxxx8 as odrv0                                                                  Website: https://odriverobotics.com/                                                                       Docs: https://docs.odriverobotics.com/                                                                     Forums: https://discourse.odriverobotics.com/                                                              Discord: https://discord.gg/k3ZZ3mS                                                                        Github: https://github.com/madcowswe/ODrive/                                                                                                                                                                          

Please connect your ODrive.                                                                                You can also type help() or quit().                                                                                                                                                                                   

In [1]: odrv0.save_configuration()                                                                         Oh no odrv0 disappeared                                                                                                                                                                                               

Reconnected to ODrive 2xxxxxxxxxx8 as odrv0                                                                In [2]:   
```

samuelsadok, 2021-09-03:
This is normal behavior. Saving configuration is followed by an automatic reboot because it can halt the CPU for a short while, potentially resulting in missed encoder steps.

closed 2021-09-03 by samuelsadok

mirano-Pat, 2021-09-03:
@samuelsadok 
Thanks. Thank you for your help !