odriverobotics/ODrive #604: ILLEGAL_HALL_STATE when ODrive recognizes state changes

Richard-Haes-Ellis, 2021-09-10:
**Describe the bug**
I had two hoverboard motors with hall sensors working with ODrive V3.5-56V Firmware 0.4.12 working fine untill updated to version 0.5.3.

The M0 channel fails to recognize the hall sensor signals, the states are recognized fine as I can plot them and are identical to channel M1 which works fine: 

Channel M0 hall state plot:

[image]

Channel M0 shadow_count does not recognize the pulses and stays at zero.

So when calibrating for polarity:
`
odrv0.axis0.requested_state = AXIS_STATE_ENCODER_HALL_POLARITY_CALIBRATION`

It fails with code 16: ENCODER_ERROR_ILLEGAL_HALL_STATE, but as you can see clearly in the plot It never goes to an illegal hall state. 

[image]

[image]

**To Reproduce**

Configure any hoverboard motor with hall sensors on channel M0 with ODrive 3.6-56V with firmware 0.5.3

**Expected behavior**

Should recognize the states and finish calibrating successfully.

**Desktop (please complete the following information):**

 - OS: Ubuntu 20.04
 - odrivetool Version : ODrive control utility v0.5.3

**Additional context**

Adding capacitors yielded the same exact results.
Swapping the motors yielded the same exact results. 
Only having one motor connected yielded the same exact results.
I've measured the pulses with a multimeter and can see clearly the states changing in the correct order and correct voltage.

Wetmelon, 2021-09-13:
Workaround:  Disable the hall polarity check with `encoder.config.hall_polarity_calibrated = True`.  

* If you have a standard 120deg polarity output, set `encoder.config.hall_polarity = 0`.
* Valid settings for `hall_polarity` are `0b000`, `0b001`, `0b010`, and `0b100`, corresponding to the sensor you wish to flip.

Richard-Haes-Ellis, 2021-09-13:
This worked! Still got ILLEGAL_HAL_STATE_ERROR when doing offset calibration but calibrated values were correct so I also bypassed it too and was able to go into closed loop control. 
Many thanks!

closed 2021-09-13 by Richard-Haes-Ellis