odriverobotics/ODrive #642: Native Protocol: sequence number bug

dakejahl, 2021-11-10:
Certain sequence numbers cause the odrive to fail to respond to a request properly. If I keep the sequence numbers between 0 and 5 I have no issues. I will probably just do this since the sequence numbers don't seem to serve any purpose anyway, but I wanted to bring it up.

**Problem numbers until I got bored of testing**
10
269
298
299
300
301
302
303
525
526

Jascha-N, 2021-12-09:
May be related to #573? In v0.5.1 many native protocol commands get interpreted as ASCII commands by the ODrive. And in my experience this is not entirely random and has to do with the contents of the command, including the sequence number. Disabling the ASCII protocol in the code fixes the issue in v0.5.1. Upgrading to a newer version fixes this issue too but for me brought a myriad of other issues and I had to fall back to using the ASCII protocol only.

samuelsadok, 2021-12-14:
Yeah a conflict with the ASCII is a likely cause for this.

Official Fibre code avoids this conflict by hardcoding the very first bit in the packet to 1: https://github.com/odriverobotics/ODrive/blob/master/Firmware/fibre-cpp/legacy_protocol.cpp#L208

Later versions of the firmware however don't allow the ASCII-protocol and Fibre to be enabled at the same time on the same channel anymore.

@dakejahl can you check if this problem persists on the latest firmware? Otherwise we can close this issue.

dakejahl, 2021-12-23:
>Upgrading to a newer version fixes this issue too but for me brought a myriad of other issues and I had to fall back to using the ASCII protocol only.

When I tried the latest firmware (0.5.4) the native protocol over serial did not work at all. I am unsure if this was due to my test setup being incorrect or if it's a problem with firmware. Once I test 0.5.4 again I will let you know. I will close this for now since this is indeed a duplicate of #573

closed 2021-12-23 by dakejahl

samuelsadok, 2022-01-05:
Note that on 0.5.4 you need to enable the native protocol on UART with `odrv0.uart0_protocol = StreamProtocolType.FIBRE`.