odriverobotics/ODrive #643: Static analysis report

vChavezB, 2021-11-16:
I ran a static analysis tool (PVS Studio) to find possible warnings or errors and found the following:

1) Variable [active_ ](https://github.com/odriverobotics/ODrive/blob/0121ccf2490f5f0664d26ac248601472157b5ee7/Firmware/MotorControl/acim_estimator.hpp#L25) of type float in class AcimEstimator is treated as boolean

https://github.com/odriverobotics/ODrive/blob/0121ccf2490f5f0664d26ac248601472157b5ee7/Firmware/MotorControl/acim_estimator.cpp#L11

2) The following condition is always false

https://github.com/odriverobotics/ODrive/blob/c58aeee6829bd3246343aff04af43ab0dbd52e6f/Firmware/MotorControl/axis.cpp#L484

3) The value '0xffffffff' is out of range of enum values. This causes unspecified or undefined behavior. 

https://github.com/odriverobotics/ODrive/blob/c58aeee6829bd3246343aff04af43ab0dbd52e6f/Firmware/Drivers/DRV8301/drv8301.cpp#L134

4) Following cases perform the same actions

https://github.com/odriverobotics/ODrive/blob/c58aeee6829bd3246343aff04af43ab0dbd52e6f/Firmware/MotorControl/encoder.cpp#L587-L596

These are some of the ones I found interesting and wanted to share this info in case it is useful

Regards

Wetmelon, 2021-11-16:
Thanks for the report.  PVS Studio is reputable. 

1. Yep good catch.
2. Hmm... I'm not sure that's right.  If anything, it should always be _true_
3. It is not out of range of int32_t (default enum type), but it's possible that PVS Studio is picking up that the compiler is making the enum type smaller than that.  A better warning would be that the enum's base type is unspecified, which we can set.
4. valid, but I think we'll deliberately leave those separate.

Static analysis in general is quite useful.  I have CppCheck running in my IDE but we don't have it in CI as of yet.  Thanks for reminding us to modernize ;)

vChavezB, 2021-11-17:
Oh BTW about #2. I think the issue arises since the following line checks for the same state:

https://github.com/odriverobotics/ODrive/blob/c58aeee6829bd3246343aff04af43ab0dbd52e6f/Firmware/MotorControl/axis.cpp#L462

which in turn makes the nested if with the same condition false as it was already checked before.

And good to know you intend to add it to your ci in.the future :)

commit 86ff244b (2021-11-16): Fix a couple static analysis catches

Fixes #643
closed 2021-11-20 by Wetmelon