odriverobotics/ODrive #647: INPUT_MODE_VEL_RAMP applies to velocity setpoint instead of actual velocity

szeder, 2021-12-04:
**Describe the bug**
Quoting `Firmware/odrive-interface.yaml`, `INPUT_MODE_VEL_RAMP` "Ramps a velocity command from the current value to the target value", not the actual velocity.
This can cause sudden velocity jumps, as shown in the middle of the blue plot on the attached image. 
Furthermore, it can cause delayed reaction to a setpoint change, which might be considered a safety issue.

[image]

**To Reproduce**
The above plot was generated by copy-pasting the following commands to `odrivetool`:
```python
import time
odrv0.axis1.controller.config.control_mode = CONTROL_MODE_VELOCITY_CONTROL
odrv0.axis1.controller.config.input_mode = INPUT_MODE_VEL_RAMP
odrv0.axis1.controller.config.vel_limit = 1
odrv0.axis1.controller.config.vel_limit_tolerance = 3
odrv0.axis1.controller.config.vel_ramp_rate = 1
odrv0.axis1.controller.input_vel = 2
# First `vel_estimate` and `vel_setpoint` both ramp up to 1, and then
# `vel_setpoint` continues to increase to 2 while `vel_estimate` stays
# around 1.
time.sleep(3)
odrv0.axis1.controller.config.vel_limit = 2
# Sudden velocity change, `vel_estimate` now jumps right up to 2
# without ramping.
time.sleep(1)
odrv0.axis1.controller.config.vel_limit = 1
# Sudden velocity change, `vel_estimate` jumps down to 1.
time.sleep(1)
odrv0.axis1.controller.input_vel = 0
# No velocity change for a while: `vel_setpoint` starts to ramp down,
# but as long as it is above 1 `vel_estimate` doesn't change.
```
**Expected behavior**
Arguably `INPUT_MODE_VEL_RAMP` works as documented, but I expected from a velocity ramping feature that it avoids all sudden velocity changes without introducing any delays.  In particular I expected smooth ramps up and down instead of sudden velocity changes in the middle of the plot, and immediate velocity reduction to the `input_vel = 0` command. 

**Desktop (please complete the following information):**
 - odrivetool Version (`odrivetool --version`): v0.5.4

samuelsadok, 2021-12-17:
Thanks for bringing this up. We put a fix in our next-generation firmware for this. With this fix, when you increase the velocity limit, the setpoint doesn't jump up anymore but observes the `vel_ramp_rate`. However when you decrease the velocity limit, the absolute setpoint limit takes precedence over the setpoint rate limit, so in this case the setpoint will still snap to the lower limit.

szeder, 2022-09-22:
Could you please point me to the commit in the current firmware fixing this issue?

Wetmelon, 2022-09-22:
Hi @szeder, this was not planned to be fixed.  But actually the Pro firmware fix looks fine for legacy also.  Please try

closed 2022-09-22 by Wetmelon