odriverobotics/ODrive #648: ERROR_UNBALANCED_PHASES triggered for some motors

pbrier, 2021-12-08:
**Describe the bug**
The phase unbalance check prevents enabling some motors. (causing ERROR_UNBALANCED_PHASES).

**To Reproduce**
Take a motor with a slight unbalance in the windings (e.g. 10%) (and a  high inductance?), and it "almost always" fails to enable. This behavior is different comprared to older firmware versions that did not have the check. The motor is performing within its specifications.

**Expected behavior**
No failure enabling the drive, unless the wires are disconnected, shorted, or there is another significant connection or motor problem.

**Desktop (please complete the following information):**
odrive master (0.5.4-dev), linux, serial connection

**Additional context**
The older firmware did not check the unbalance. The current unbalance check puts a (arbitrary?) 10% limit on the unbalance.  In addition, I suspect there is a timing  or rotor position dependency on the check: I have the impression that with the motor I have the some initial positions makes it pass the test "sometimes" The motor is a high-pole wheel hub motor (110mm diam).

Raising the unbalance limit to 15% makes it pass the test "always", and I made it 20% to be safe on my setup to fix this:

```./Firmware/MotorControl/motor.cpp#L447``` was changed to :

[```if (is_nan(I_beta) || (abs(I_beta) / test_current) > 0.1f) { ``` ](https://github.com/odriverobotics/ODrive/blob/16801640ee84dede2c16c928c886061007f67a9a/Firmware/MotorControl/motor.cpp#L447])

Was changed to:

```if (is_nan(I_beta) || (abs(I_beta) / test_current) > 0.2f) {```

to fix this.

**Possible fixes:**

1) Set the hardcoded limit higher (e.g. 20%) by default
2) Make the limit configurable (e.g. ```motor.config.maximum_unbalance=10``` [%])
3) Add some additional time to let the current stabilize (100msec?, configurable?)
4) Experiment with the influence of the test current for each motor and setup (cumbersome)
5) Buy a better motor (not an option for me)

Jascha-N, 2021-12-08:
I have the same issue with an official motor sold on the ODrive website (D6374). Used to work fine on older firmware.

samuelsadok, 2021-12-10:
Thanks for the report!

Yeah the hardcoded limit is somewhat arbitrarily chosen. The most common reason for the error is an actual bad connection on the motor phases (e.g. screws not tightened). But I guess it's possible for some motors to legitimately have unbalanced phases.

I raised to limit to 20%, I think that is the easiest sensible solution.

Note that the resistance measurement already holds the current for around 1 second (the silent time between starting motor calibration and hearing the motor beep) and has an I-gain that should smooth out most measurement noise.

closed 2021-12-10 by samuelsadok