odriverobotics/ODrive #657: #please connect your Odrive

HusamZain, 2022-01-31:
I got a message when  open odrivetool from terminal , I'm using ubuntu 20.04 
it say's '' please connect your odrive'', i'm connecting the odrive to my pc via USB, 
and when i use 'lsusb' command it seems it doesn't show the odrive at all.

does anyone know how to fix this issue?

samuelsadok, 2022-02-02:
> it seems the USB is no longer detectable

Does that mean it doesn't show up in `lsusb` anymore? When you put it into DFU mode the name in `lsusb` should be something like "STM32 Bootloader".

If it still shows in `lsusb` in DFU mode, you might be able to recover it using `odrivetool dfu`.

If it doesn't show in `lsusb` at all, it's possible that the MCU got fried somehow. If you think this could be a manufacturing issue please email info@odriverobotics.com with a link to this thread and the following info:
 - Your order number
 - A brief description/list/picture of what you have wired up to the ODrive (power supply, motor, encoder, other external MCUs, ...)
 - A description of what the ODrive was doing at the time when the issue appeared

Wetmelon, 2022-02-16:
Please post to the forum or ask on the discord channel if this is still an issue.

closed 2022-02-16 by Wetmelon