odriverobotics/ODrive #668: Please add PWM mode: PWM+- or  PWM+ DIR or PWM one wire 0-50-100

active-game, 2022-02-12:
I'm trying to make a steering wheel using MMOS or EMC type controllers, they all output a PWM signal: PWM+- or PWM+ DIR or PWM one wire 0-50-100 , but none of these PWM signals support odrive !
Please add this feature, I will be very grateful!

Wetmelon, 2022-02-14:
Can you be more specific please?  What is PWM +-?

active-game, 2022-02-14:
> Can you be more specific please? What is PWM +-?

I mean Left and Right channels 2 wire PWM like on BTS7960

madcowswe, 2022-02-16:
My interpretation is the following:
> PWM+-
two wires: pwm 0-100% to go forward, pwm the other signal to go backwards
> PWM+ DIR
0-100% PWM and a digital direction signal
> 0-50-100
0% pwm max speed one way, 50% PWM stop, 100% pwm max speed the other way.

active-game, 2022-02-16:
> Моя интерпретация следующая:
> 
> > ШИМ+-
> 
> два провода: pwm 0-100% для перехода вперед, pwm другой сигнал для перехода назад
> 
> > ШИМ+ НАПРАВЛ.
> 
> 0-100% PWM и цифровой сигнал направления
> 
> > 0-50-100
> 
> 0% максимальной скорости PWM в одну сторону, 50% остановки PWM, 100% максимальной скорости PWM в другую сторону.

Yes, everything is right. Thank you !
It is strange that the controller created for robotics does not have industrial PWM, but instead only RC PWM

active-game, 2022-03-05:
No news?

Wetmelon, 2022-03-06:
> No news?

We have no plans to add this features at this time.

PWM 0 - 50% - 100% one-wire mode is supported as part of the RC PWM mode

closed 2022-09-21 by Wetmelon