odriverobotics/ODrive #674: ERROR_ILLEGAL_HALL_STATE firmware bug?

blauertee, 2022-02-21:
I found Multiple instances of people asking about this in the Odrive Forums and nobody was able to offer a real solution so far except for downgrading the firmware to 0.5.1. 

Nobody really disagreed with there being a bug in the firmware on the other hand configuring hall mode just seams to be tricky ([see](https://discourse.odriverobotics.com/t/solved-the-hall-sensor-pull-up-resistor-is-strange/1994) sometimes.

Is there a bug in the firmware?

**Describe the bug**
Well described [here](https://discourse.odriverobotics.com/t/error-in-firmware-shadow-count/7411/3)

**Expected behavior**
Calibrate hall sensors

**Desktop (please complete the following information):**
 - Ubuntu
 - odrivetool Version (`odrivetool --version`)

Thanks in Advance, Jakob

samuelsadok, 2022-04-14:
Sorry for letting this sit so long. Could you provide the (minimal) sequence of `odrivetool` commands that you ran, starting at `odrv.erase_configuration()` and leading up to the unexpected behavior?

blauertee, 2022-04-14:
Since I do not work for the company anymore for which I was working with odrive. I'm not really interested in solving this anymore. So I'll just close this feel free to reopen, if you face similar problems.

closed 2022-04-14 by blauertee