odriverobotics/ODrive #675: Incoherent Seral Number Type

blauertee, 2022-02-22:
**Describe the bug**
`odrv0.serial_number` is an int whereas in `discover_any(serial_number)` the serial number is expected as a hex string. I think this should be coherent. Somewhere there should be a line like `int(serial_number, 16)` or `hex(serial_number)` depending on what you prefer.

samuelsadok, 2022-02-22:
The string variant is the correct one. However the Fibre protocol currently does not support string properties. Therefore fixing this would either be fairly hacky or non-trivial.

blauertee, 2022-02-22:
Okay that makes sense, maybe name it `serial_int` then? Add a `odrive.dump_serial_nr(odrv0)` method. Anyway I guess it's not that pressing.

samuelsadok, 2022-02-25:
Well `odrive.dump_serial_nr(odrv0)` would be equivalent to `hex(odrv0.serial_number)`. But yeah it's not a huge priority at the moment.

closed 2022-09-21 by Wetmelon