odriverobotics/ODrive #688: Drifting in Position Mode

tobbelobb, 2022-05-03:
**Describe the bug**
The motor doesn't keep its position as it should.

**To Reproduce**
 - Climb up to your Hangprinter ceiling unit and release the belts from the motors for the 10000'th time.
 - Make sure it uses ODrive v3.6, 24V and the [330 kv motors](https://odriverobotics.com/shop/dual-shaft-motor-d5212s-300kv)
 - Build ODrivefirmware at commit daa089c3 and flash it.
 - Run odrivetool from the repo: `$ ~/repos/ODrive/tools/odrivetool`
 - `In [1]: odrv0.erase_configuration()`
 - `In [2]: exit`
 - `$ ~/repos/ODrive/tools/odrivetool restore-config odrive-config-AB.json`, where odrive-config-AB.json is [linked here](https://gitlab.com/tobben/hangprinter/-/blob/9c4e356ceeecb03f1806753a62d11261f0706e89/firmware/ODrive/odrive-config-AB.json)
 - `$ ~/repos/ODrive/tools/odrivetool`
 - `In [1]: odrv0.config.brake_resistance = your_measured_value`
 - ```html
        odrv0.axis0.requested_state = AXIS_STATE_IDLE
        odrv0.axis1.requested_state = AXIS_STATE_IDLE
        odrv0.save_configuration()
        odrv0.axis0.requested_state = AXIS_STATE_FULL_CALIBRATION_SEQUENCE
        odrv0.axis1.requested_state = AXIS_STATE_FULL_CALIBRATION_SEQUENCE
        odrv0.axis0.motor.config.pre_calibrated = True
        odrv0.axis1.motor.config.pre_calibrated = True
        odrv0.axis0.encoder.config.pre_calibrated = True
        odrv0.axis1.encoder.config.pre_calibrated = True
        odrv0.axis0.requested_state = AXIS_STATE_IDLE
        odrv0.axis1.requested_state = AXIS_STATE_IDLE
        odrv0.save_configuration()```
 - Climb to ceiling and re-attach belt to motor0.
 - Put a slight constant load on motor0. This is not strictly neccesary, the bug appears without it as well, but the load makes the drift much larger.
 - Observe and note down motor0's `pos_estimate`. For me it was 0.0057.
 - While under slight load, move motor0 backwards ca 1 revolution with step/dir signals. For me this moved `pos_setpoint` from 0.899 to 99.83.
 - Still under slight load, step motor0 forwards again by the exact same amount. For me this mover `pos_setpoint` from 99.83 to 0.899 as expected.
 - Observe your `pos_estimate` again. For me it was 0.03457.

**Expected behavior**
I expected the `pos_estimate` to be the same as before (0.0057) after I stepped `pos_setpoint` back to its original value.

Instead, the motor has drifted away from it's initial position by ca 1.65 degrees, according to the new `pos_estimate` (0.03457).

The drift in `pos_estimate` is also observed in physical reality. During a real 3d-print with Hangprinter, similar drift happens several times per second, and the print quickly go haywire. When using 0.5.1 this doesn't happen.

I think the drift happens as the moment when we're crossing down through `pos_setpoint = 0`

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - odrivetool Version v0.5.4.dev (commit daa089c3)

**Additional context**
I was asked on Discord if this happens even if I disconnect the step signal and hard-wire it to GND or 3.3V.
I can't test this, since I can't move the motor any other wan than with the step/dir signals.
The ODrive in the current configuration doesn't respond properly to `input_pos` changes.
I tried

`In [56]: odrv0.axis1.controller.input_pos = 0.1`

... but motor remained still, and ` odrv0.axis1.controller.input_pos` kept its old value.

I strongly doubt step/dir signal noise being the problem since the machine has been running tens of hours using 0.5.1 with no drift, and shows substantial drift with v0.5.5 (devel) within seconds of starting a print.

Wetmelon, 2022-09-21:
Did #680 fix this issue?

Wetmelon, 2022-09-26:
> We found a discrepancy between counters two counters: axis0.encoder.shadow_count and axis0.encoder.count_in_cpr.

Doesn't sound right.  Only way that happens is if you're using an absolute encoder.