odriverobotics/ODrive #691: Set Velocity Ramp Rate via CAN Simple

lekoook, 2022-05-11:
**Describe the solution you'd like**
Additional CAN Simple message that can alter the velocity ramp rate (ODrive.Controller.Config.vel_ramp_rate).

**Additional context**
I am also willing to contribute to make this happen.

samuelsadok, 2022-08-11:
Can you give a brief description about your use case for this CAN message? In particular, is it necessary to change this variable at runtime (while the machine is in operation in the field)?

Note that CANSimple is not intended to replace one-time setup via USB because manually adding a message for every configuration variable would not be maintainable.

Regarding implementation, #699 could serve as an orientation (the diff contains some noise, the relevant changes are only one commit).

Wetmelon, 2022-08-19:
#707 Is the updated one

lekoook, 2022-09-22:
> Can you give a brief description about your use case for this CAN message? In particular, is it necessary to change this variable at runtime (while the machine is in operation in the field)?
>
> Note [...]

The use case we have would be to change the ramp rate depending on the user's acceleration profile preference. This will not be a one-time setup as the user can change the "acceleration feel" as and when they like.

samuelsadok, 2023-11-06:
Sorry forgot to follow up here. This is now possible on ODrive Pro/S1 using [Arbitrary Parameter Access](https://docs.odriverobotics.com/v/latest/guides/can-guide.html#arbitrary-parameter-access).