odriverobotics/ODrive #700: No axis error via CAN heartbeat when motor is disconnected

falkoschindler, 2022-06-29:
**Describe the bug**
When the motor is disconnected from the ODrive (motor and encoder), the CAN interface still sends heartbeat messages with "axis error" 0. The only indicator that something happened is the "axis current state" changing from "closed loop" to "idle". Meanwhile, the ODriveTool yields multiple errors:

```
In [7]: dump_errors(odrv0)
system: no error
axis0
  axis: no error
  motor: Error(s):
    MotorError.UNKNOWN_TORQUE
    MotorError.UNKNOWN_VOLTAGE_COMMAND
  DRV fault: none
  sensorless_estimator: no error
  encoder: Error(s):
    EncoderError.ILLEGAL_HALL_STATE
  controller: Error(s):
    ControllerError.INVALID_ESTIMATE
axis1
  axis: no error
  motor: no error
  DRV fault: none
  sensorless estimator: no error
  encoder: no error
  controller: no error
```

**To Reproduce**
Bring an axis into "closed loop" state.
Unplug the motor including its encoder from the ODrive while listening to the CAN heartbeat.

**Expected behavior**
I would expect a non-zero axis error in the CAN heartbeat message.

**Desktop**
 - OS: Ubuntu 18.04.5 LTS
 - odrivetool Version: 0.6.1

**Additional context**
We use a single cable with a 9-pin connector for connecting motor and encoder to the ODrive.

samuelsadok, 2022-08-11:
Were you trying with firmware v0.5.4 or with the latest `devel`? I believe on `devel` this issue should already be fixed as we added back the `AXIS_ERROR_MOTOR_FAILED` flag.

samuelsadok, 2022-08-11:
Ok you can try with v0.5.5 then which got released just now

Wetmelon, 2022-09-21:
@falkoschindler Are you still seeing this issue on 0.5.5?

closed 2022-09-22 by Wetmelon

falkoschindler, 2022-09-22:
@Wetmelon Sorry, I haven't had time to look in to 0.5.5, yet.