odriverobotics/ODrive #704: Position control keeps running when the mechanical brake is engaged

Duck-Pilot, 2022-07-27:
**Describe the bug**
When engaging the mechanical brake while in closed loop position control mode, the controller keeps trying to correct motor position, but the mechanical brake makes this impossible. If the axis is moved by a large force while the brake is engaged, the controller will push a large current through the motor, but the motor can not move, so it will start heating up. 

**To Reproduce**
Set the controller to closed loop position control mode and engage the mechanical brake connected to the controlled motor, then move the motor by hand while the brake is still engaged. The controller will now start pushing current through the motor, trying to move it.
This issue might also present itself if the position setpoint is changed while the mechanical brake is still engaged.

**Expected behavior**
When the mechanical brake is engaged by it's command through the odrive, the controller stops controlling the motor and pushing current through it

**Desktop:**
 - OS: Windows 10
 - odrivetool v0.6.1.post0
 - odrive v3.6 56V with 5650 odrive motor and CUI AMT102 encoder
 - odrive fw version 0.5.4

**Additional context**
The motor is connected to a large rotating table through belt reductions, the encoder is connected to the motor but the brake brakes the table directly. Due to the size of the table, it is possible to move the table by a large force while the brake is engaged.

samuelsadok, 2022-08-11:
The mechanical brake engages automatically when going to IDLE state. Would it be viable for you to use a state change to IDLE instead of calling `mechanical_brake.engage()`?

Wetmelon, 2022-09-21:
I started implementing a fix for this, but it's too sketchy.  Please just drop into IDLE when you want the brake to engage.

closed 2022-09-21 by Wetmelon