odriverobotics/ODrive #708: Brushed motor control

rowanG077, 2022-09-05:
**Is your feature request related to a problem? Please describe.**
I have an application where I test BLDC motor against each other using odrive. Now I would like to start doing something similar with brushed motors. I really don't want to rewrite and split our software to a different motor controller to get brushed motors working. I known there used to be someone (or multiple?) people working on it a 2 years ago but  I can't find anything on here anymore. 

**Describe the solution you'd like**
I would like to see brushed motor supported. I'm willing to get my hand dirty myself but I probably need some pointers because I have never written a motor control. A few years ago I implemented motor thermistor support so I'm at least a little bit familiar with the code.

samuelsadok, 2022-09-09:
I think the earlier work you're referring to is https://github.com/odriverobotics/ODrive/pull/362. You can check my comment on there for some pointers.

However the required changes would be quite involved and deep and our testing capacity/capability for ODrive v3 is quite limited so unfortunately I don't think we'd be able to accept such a changeset.

For our newer generation products brushed motor support might be a thing some day but it's fairly low priority currently.

lawrence-jeff, 2022-09-09:
Changing out the control board of a commercial laser that uses brushed motors with encoders - also came here looking for an open source replacement driver that takes step/direction signals. I don't think I have the skill to help update your code but just wanted to add a vote for the feature