odriverobotics/ODrive #722: How to  manage several Odrives with UART?

active-game, 2022-10-31:
How i can  manage several Odrives with UART?
How to communicate with several Odrives in UART?
how to change address of each device without using serial number?
For example Now command : p 0 or 1 -2 0 0
I can only select the motor, how do I access another drive?

samuelsadok, 2022-11-01:
Multiple ODrives on a single UART line are not supported, you need to connect each ODrive to a separate UART port.

closed 2022-11-01 by samuelsadok

active-game, 2022-11-01:
> Multiple ODrives on a single UART line are not supported, you need to connect each ODrive to a separate UART port.

It's a pity! didn't the developers really think about adding the device address?! as it is done in others Servodrivers or VFD.

samuelsadok, 2022-11-01:
UART is usually only used for one-to-one connections. If you want to use multiple devices on one bus, I recommend to use CAN.

active-game, 2022-11-03:
> UART is usually only used for one-to-one connections. If you want to use multiple devices on one bus, I recommend to use CAN.

Thank you!  How i can use multiple  odrive in can?