odriverobotics/ODrive #731: AttributeError: module 'fibre' has no attribute 'Event'

ZhangzrJerry, 2023-02-12:
**Describe the bug**
when I run GUI/server/odrive_server.py it will show me 
```
args from python: []
Traceback (most recent call last):
  File "/home/zhangzrjerry/Documents/平衡轮/ODrive/GUI/server/odrive_server.py", line 296, in <module>
    shutdown = fibre.Event()
AttributeError: module 'fibre' has no attribute 'Event'
```

**To Reproduce**
when I type `python3 GUI/server/odrive_server.py` in terminal

**Expected behavior**
program crashed in this line
https://github.com/odriverobotics/ODrive/blob/e89d71a7d70b6415a3bdcaacc6a90a1a21f26105/GUI/server/odrive_server.py#L296

**Desktop (please complete the following information):**
 - OS: ubuntu22.04.1LTS
 - odrivetool Version latest

samuelsadok, 2023-03-03:
This version of the GUI is unfortunately no longer actively supported, but if you make a PR with a small fix I could look at it.

Reno43, 2023-04-04:
Is there a version of GUI is supported for Odrive V3.6?

samuelsadok, 2023-04-04:
The GUI in this repository would be compatible with ODrive v3.6 if you can get it to run. However this version is no longer maintained and apparently has issues with starting, according to the OP. There may also be unofficial community projects around.

However if your application allows, we strongly recommend to use the newer generation ODrive Pro/S1 for several reasons. They are both supported by our [new GUI](https://gui.odriverobotics.com/configuration).

closed 2023-08-19 by ZhangzrJerry