odriverobotics/ODrive #737: Issues with [USB] Could not open USB device: -3[0m

hpark416, 2023-06-25:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Run the example code, no power provided to the motors just yet and then run the code again and the error output should mention how USB could not open USB device

**Expected behavior**
Expected it to terminate the drive instance? if it fails calibration but it seems to just close out and leave it in a perpetual state of initializing.

**Desktop (please complete the following information):**
 - OS: [Windows 10]
 - odrivetool Version (`ODrive control utility v0.6.5.post2`)

**Additional context**
Have been getting around to learning how to use the API and making a Python file for getting started; I noticed that if you ask it to do calibration and power is not provided it can get stuck in [Initialization]

Is there a convenient way to like to terminate that state I guess; If I cycle the odrive's power the problem persists indicating it must be like the system is probably having an issue with not having closed the call in the initial call to do a calibration(initialization)

Workarounds for this as stated is just cycling my PC but being new I feel like maybe I don't know if there is a method already that exists but as I said, pretty new to coding so just trying to see if anyone can enlighten me as I continue to search around.

Here is the error when you try to reconnect to the odrive tool:
"c:\Users\Richard\Desktop\Humanoid Robot SW\RobotStartup.py"
finding an odrive...
[91;1m16:03:42.866207200 [USB] Could not open USB device: -3[0m

So I know the program gets caught up on this line "my_drive = odrive.find_any()"
Not sure why this happens, and kinda don't wanna have to keep restarting my PC.

samuelsadok, 2023-06-26:
I'm not sure why rebooting the PC makes a difference. When you close odrivetool (or your Python script) then the OS will release any USB resources, and when you restart the ODrive it will also reset any communication state.
Maybe the Python instance didn't fully close?
Does this also happen when you use odrivetool?

Which ODrive and firmware version is this?

hpark416, 2023-06-26:
Yeah, I suspect it had to do with it not terminating memory. I did find some strange way to "reset" it though... 

Yes once it cannot reconnect in Python throwing the error, I could not get connected in any method (using terminal and odrivetool, using a python file with the import for odirve, and even the web GUI could not see the USB devices anymore when plugged in leading me to do point 2 below after which things became a little more salvageable between faults)

Odrive Pro Drivers should be the latest version since I had to do a pip install on that system that day.

Here are some additional findings in case this helps with updates:
1. Dont let it initialize without power when running without the GUI or else it gets stuck in INITIALIZING
2. I tried to use something called Zadig based on this forum: https://discourse.odriverobotics.com/t/no-usb-device-showing/1198. Not sure if that helped because I'm not too savvy with the whole whatever it's doing.
3. At some point I managed to utilize both the terminal and the python terminal in vscode to test quickly and reset if I need to.
4. You can have the web GUI open after it faults to clear the error and observe the last known status of the drives. But cannot access the drivers if GUI is still open so before running any Python stuff close the GUI and it can be recognized again. I would just check after a run with the GUI hit clear errors a couple of times(POINT 5 is the major one tho
5. It Seems that if you can get a connection to the drivers again, and can call full calibration (this is in python), and then stop the code before it finishes calibration the drivers seem to have snapped out of the issue.

Tips: 
Make sure to utilize the code run in the terminal in vs code, just more convenient
Hopefully, this helps I just am writing this in case others are finding this issue.

samuelsadok, 2023-08-11:
Thanks for the writeup.
I would like to add for clarification: we generally do not recommend to use Zadig anymore, except for DFU mode. On the contrary, when there's suspicion of driver problems, I would usually first ask to _uninstall_ Zadig drivers. The reason is that since firmware v0.6.4, the ODrive tells Windows to load the correct driver automatically. If you use Zadig, it can interfere with this automatic setup and it's a bit convoluted to undo what Zadig does (uninstall instructions [here](https://docs.odriverobotics.com/v/latest/troubleshooting/troubleshooting.html#uninstall-zadig-drivers-windows-only)).

closed 2023-08-11 by samuelsadok