odriverobotics/ODrive #738: axisError = can.Signal("Axis_Error", 0, 32, receivers=['Master'], choices={error.value: error.name for error in AxisError})

zhangbeyonder, 2023-07-19:
Hi, Guys. When I `make` the master branch, some errors occured. An error report:
```
Traceback (most recent call last):
  File "D:\WorkSpace\ODrive\tools\create_can_dbc.py", line 15, in <module>
    axisError = can.Signal("Axis_Error", 0, 32, receivers=['Master'], choices={error.value: error.name for error in AxisError})    
TypeError: __init__() got an unexpected keyword argument 'choices'
make: *** [Makefile:36: all] Error 1

```
Do you know how to fix it?

**Desktop **
 - OS: Windows 10
 - odrivetool Version (`odrive-0.6.7`)
 - I am in this branch:commit 58fdd3fdfdcaeff76547870f5e7acdf1673479e8 (HEAD, origin/master, origin/HEAD, master)
Author: samuelsadok <samuel.sadok@bluewin.ch>
Date:   Tue May 9 12:18:21 2023 +0200

    Update CHANGELOG.md

samuelsadok, 2023-07-19:
There was a recent breaking change in the `cantools` Python package. As a workaround you can install an older version with `pip install "cantools<39.0.0"`.

(your `pip` command may look a bit differently, depending on your Python installation)

Alternatively, you can try to make our code compatible with the newer cantools version and open a PR with the fix.

zhangbeyonder, 2023-07-26:
Thank you for your help. The command does make sense. 
BTW, I am a beginner in ODrive. I would like to try to fix it and it may need some time.

closed 2023-07-26 by zhangbeyonder

1256879347, 2023-11-10:
I have the same problem. Have you solved it? How to do