odriverobotics/ODrive #739: Torque_msgs_ms rate from the gui is not setting the torque message

r4hul77, 2023-08-29:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps and configuration necessary to reproduce the behavior.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows 10]
 - odrivetool Version (`odrivetool --version`)

**Additional context**
Add any other context about the problem here.

samuelsadok, 2023-08-30:
Does the rate setting work for other messages?

Can you copy&paste the diagnostics info from the GUI?

[image]

r4hul77, 2024-01-11:
Yes the rate settings work for other messages, I'm reading the message through CAN and able to control the motor. I want all the messages at 10ms but it appears I'm missing torque messages always

samuelsadok, 2024-01-16:
What's your firmware version? (see previous message to copy diagnostics info)

Do you have a way to log CAN messages? For instance on Linux you can install python-can (`pip3 install python-can`) and run `can_logger -i socketcan -c can0`. Could you grab 1 seconds worth of message log and share that?

r4hul77, 2024-03-22:
The latest update fixed this issue if anyone is here with a similar problem

closed 2024-03-22 by r4hul77