odriverobotics/ODrive #740: Python ImportError when compiling Odrive Firmware

ChnTamap, 2023-11-17:
Hi,
I am encountering a Python error while compiling the firmware

**Describe the bug**
```
PS E:\Project\Libraries\Odrive\Odrive\Firmware> make
board version: v3.6-56V
Firmware version 0.5.6-dev (fw-v0.5.6-1-g58fdd3fd*)
 [     ] 100%
Traceback (most recent call last):
  File "E:\Project\Libraries\Odrive\Odrive\tools\create_can_dbc.py", line 1, in <module>
    from cantools.database import *
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python39\lib\site-packages\cantools\__init__.py", line 9, in <module>
    from . import j1939, logreader, tester
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python39\lib\site-packages\cantools\tester.py", line 8, in <module>
    import can
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python39\lib\site-packages\can\__init__.py", line 81, in <module>
    from . import broadcastmanager, interface
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python39\lib\site-packages\can\interface.py", line 12, in <module>
    from .bus import BusABC
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python39\lib\site-packages\can\bus.py", line 25, in <module>
    from typing_extensions import Self
ImportError: cannot import name 'Self' from 'typing_extensions' (C:\Users\Administrator\AppData\Local\Programs\Python\Python39\lib\site-packages\typing_extensions.py)
make: *** [Makefile:36: all] Error 1
```

**To Reproduce**
```
cd Firmware
make
```

**Desktop**
 - OS: Windows 10
 - odrivetool Version : ODrive control utility v0.6.7 (unused)

The .elf file has been successfully compiled, but I still want to ask how to resolve this issue.
My software versions: Python 3.9.5, cantools-39.4.0
Do I need to install a specific version of Python or cantools? Or do I need to modify which code?

Thank you!

samuelsadok, 2023-11-21:
From a quick search, it could be that your `typing_extensions` is too old, so try
```
pip3 install --upgrade typing_extensions
```

If that doesn't work, you can try to downgrade cantools:
```
pip3 install uninstall cantools
pip3 install "cantools<39.0.0"
```

ChnTamap, 2023-12-20:
> From a quick search, it could be that your `typing_extensions` is too old, so try
> 
> ```
> pip3 install --upgrade typing_extensions
> ```
> 
> If that doesn't work, you can try to downgrade cantools:
> 
> ```
> pip3 install uninstall cantools
> pip3 install "cantools<39.0.0"
> ```

Thank you! I upgrade the `typing_extensions`, And it works.
But a new problem has emerged:
```
Traceback (most recent call last):
  File "E:\Project\Libraries\Odrive\Odrive\tools\create_can_dbc.py", line 15, in <module>
    axisError = can.Signal("Axis_Error", 0, 32, receivers=['Master'], choices={error.value: error.name for error in AxisError})
TypeError: __init__() got an unexpected keyword argument 'choices'
make: *** [Makefile:36: all] Error 1
```
I have try to upgrade `cantools`, But it seems to have no effect.

ChnTamap, 2023-12-25:
> From a quick search, it could be that your `typing_extensions` is too old, so try
> 
> ```
> pip3 install --upgrade typing_extensions
> ```
> 
> If that doesn't work, you can try to downgrade cantools:
> 
> ```
> pip3 install uninstall cantools
> pip3 install "cantools<39.0.0"
> ```

Thank you! I upgrade the `typing_extensions`, a new problem has emerged (at `create_can_dbc.py:15`). then I downgraded `cantools` and build successed.
Thanks again.

closed 2023-12-25 by ChnTamap