odriverobotics/ODrive #746: [USB] Could not open USB device: -3

mcdominik, 2024-05-15:
Does any one encounter this error?

[image]

It show when I try to run this snippet:

[image]

My setup:
Jetson Orin 
odrive 0.6.8
python 3.8.10

samuelsadok, 2024-05-27:
This the error code -3 refers to a libusb error code, so this means [`LIBUSB_ERROR_ACCESS`](https://github.com/libusb/libusb/blob/master/libusb/libusb.h#L1292).

Did you already set up the udev rules?
See here: https://docs.odriverobotics.com/v/latest/interfaces/odrivetool.html#installation > Linux > "Set up USB device permissions."

mcdominik, 2025-01-31:
Apologies for the late response; the issue was resolved after changing the USB hub.

closed 2025-01-31 by mcdominik