odriverobotics/ODrive #749: error connecting to runtime ODrive: BindingEror: Cannot use deleted val, handle = 0

mcdominik, 2024-07-16:
**Describe the bug**
I am trying to connect motor in web GUI, but get an error:
```
error connecting to runtime ODrive: BindingEror: Cannot use deleted val, handle = 0
```
[image]

**Desktop:**
 - OS: Ubuntu 20.04, arm64 (Jetson Orin)
 - odrivetool Version: ODrive control utility v0.6.9.post0
 - Chromium browser

Unfortunately I found no information online about this particular error

bnlerner, 2024-07-26:
any resolution for this?

samuelsadok, 2024-07-26:
Sorry I thought the OP was the same person as someone who asked the same question on Discord one day later.

In that case, this error was caused by permission errors.
- If you're on Windows, have you used Zadig before? Make sure that any modifications by Zadig are undone. E.g. go to device manager and uninstall the driver for "ODrive Native Interface", then re-plug the ODrive.
- If you're on Linux, make sure you have the udev rules installed (they are shown on the splashscreen of the GUI)

If this doesn't help, please check the browser's developer console and let me know if there are more error details.

Btw note that this repo is for ODrive v3.x, so for questions regarding the GUI / newer ODrives, the recommended way is to ask on Discord or send us an email.

JurajLieskovsky, 2024-11-21:
I have encountered the same issue on Ubuntu 22.04.5 LTS. The issue persists after installing udev rules with the provided command. This is the console output

```
[UsbDiscoverer] Failed to open USB device: Promise failed: SecurityError: Failed to execute 'open' on 'USBDevice': Access denied.
main.dart.js:31700 saving device list: null
main.dart.js:31700 error connecting to runtime ODrive: BindingError: Cannot use deleted val. handle = 0
BindingError: Cannot use deleted val. handle = 0
    at throwBindingError (https://gui.odriverobotics.com/libodrive.js:1006:48)
    at Object.toValue (https://gui.odriverobotics.com/libodrive.js:2322:13)
    at __emval_call_method (https://gui.odriverobotics.com/libodrive.js:2981:25)
    at libodrive.wasm.emscripten::val emscripten::val::call<emscripten::val, int>(char const*, int&&) const::'lambda'(emscripten::internal::_EM_METHOD_CALLER*, emscripten::_EM_VAL*, emscripten::internal::_EM_DESTRUCTORS**, void const*)::operator()(emscripten::internal::_EM_METHOD_CALLER*, emscripten::_EM_VAL*, emscripten::internal::_EM_DESTRUCTORS**, void const*) const (wasm://wasm/libodrive.wasm-00d6b34a:wasm-function[4687]:0xce686)
    at libodrive.wasm.emscripten::val emscripten::val::internalCall<(emscripten::internal::EM_METHOD_CALLER_KIND)0, emscripten::val, emscripten::val emscripten::val::call<emscripten::val, int>(char const*, int&&) const::'lambda'(emscripten::internal::_EM_METHOD_CALLER*, emscripten::_EM_VAL*, emscripten::internal::_EM_DESTRUCTORS**, void const*), int>(emscripten::val emscripten::val::call<emscripten::val, int>(char const*, int&&) const::'lambda'(emscripten::internal::_EM_METHOD_CALLER*, emscripten::_EM_VAL*, emscripten::internal::_EM_DESTRUCTORS**, void const*), int&&) const (wasm://wasm/libodrive.wasm-00d6b34a:wasm-function[3771]:0xb5273)
    at libodrive.wasm.emscripten::val emscripten::val::call<emscripten::val, int>(char const*, int&&) const (wasm://wasm/libodrive.wasm-00d6b34a:wasm-function[3765]:0xb4e96)
    at libodrive.wasm.fibre::WebUsbDevice::claim_interface(int) (.resume) (wasm://wasm/libodrive.wasm-00d6b34a:wasm-function[3762]:0xb4292)
    at libodrive.wasm.UsbDevice::connect_async(fibre::Callback<void, BootloaderInfo*>, fibre::Callback<void, std::__2::basic_string<char, std::__2::char_traits<char>, std::__2::allocator<char>> const&>) (.resume) (wasm://wasm/libodrive.wasm-00d6b34a:wasm-function[2338]:0x670a9)
    at libodrive.wasm.std::__2::coroutine_handle<cppcoro::detail::task_promise<void>>::resume[abi:ne180100]() const (wasm://wasm/libodrive.wasm-00d6b34a:wasm-function[2318]:0x643c8)
    at libodrive.wasm.cppcoro::task<void>::resume() (wasm://wasm/libodrive.wasm-00d6b34a:wasm-function[2288]:0x601c0)
main.dart.js:31700 Another exception was thrown: Instance of 'minified:kc<void>'
main.dart.js:31700 saving device list: null
```

I do not encounter the same issue on my arch machine or windows machines of my colleagues.

closed 2024-11-29 by samuelsadok