odriverobotics/ODrive #750: odrv.onboard_encoder0.raw returns zero whenever odrv.rs485_encoder_group0.config set to ODRIVE_OA1

SeamusHerson-MilwaukeeTool, 2024-07-22:
The value returned by the onboard encoder appears to always be 0 when odrv.rs485_encoder_group0.config is set to ODRIVE_OA1 or AMT21_POLLING.    This results in a NO_RESPONSE error when attempting calibration.  

**To Reproduce**
In odrivetool, enter:
odrv.axis0.config.load_encoder = EncoderId.RS485_ENCODER0
odrv.rs485_encoder_group0.config.mode = Rs485EncoderMode.ODRIVE_OA1
odrv.axis0.config.commutation_encoder = EncoderId.ONBOARD_ENCODER0

odrv.onboard_encoder0.raw will always return 0, no matter the orientation of the shaft relative to the onboard encoder. 

**Expected behavior**
odrv.onboard_encoder0.raw output will change as the shaft turns relative to the encoder

**Desktop (please complete the following information):**
 - OS: windows 11
 - Originally observed in ODrive GUI
- FW 0.6.9-1

**Additional context**
Found this bug while attempting calibration with a OA1 encoder as the load encoder after a 10:1 reduction from the motor w/ the onboard encoder as the commutation encoder.

samuelsadok, 2024-07-24:
We're aware of this issue but forgot to document it properly, sorry!

The recommended workaround is:
1. Upgrade to `devel` firmware (or 0.6.10, once it's released)
2. Use `rs485_encoder_group1` instead of `rs485_encoder_group0` (and correspondingly `EncoderId.RS485_ENCODER1` instead of `EncoderId.RS485_ENCODER0`).

For others reading: this is only an issue on ODrive S1, not on ODrive Pro.

Where would be a good place in the docs where we can add this note where you would have seen it?

(Btw note that this is the repo for ODrive v3.x, the recommended way for bug reports for current-gen ODrives is via Discord or Email)

SeamusHerson-MilwaukeeTool, 2024-07-24:
Hi Sam -
Thanks for the fix - this works very well!

As far as where would be best to document this note, adding it somewhere near the "! Important" Box in Manual > Hardware Configuration > Encoders > RS485 Encoders is probably the place I would've found it most helpful.  Sorry for submitting the issue on the wrong repo - I appreciate the response regardless!

samuelsadok, 2024-07-25:
Good to know the workaround works for you!

I added the note to https://docs.odriverobotics.com/v/latest/manual/hardware-config.html#rs-485-encoders (a bit further down, but hopefully still visible).

No worries with submitting the issue here, just mentioning it in case other users read it, to keep things organized and avoid confusion.