odriverobotics/ODrive #752: ODrive v3.6 no longer "genuine" after replaced STM32

0xPIT, 2024-12-02:
Hello,

I think I got a genuine (according to the donor) ODrive v3.6 that had fried encoder inputs.
I've replaced the microcontroller and flashed the lastest firmware (ODriveFirmware_v3.6-56V.elf) using ST-Link.

Now the device is no longer "genuine" and I can't use the web gui.

Unfortunately I already threw away the defective µC to not mistake it with a good one and I was not aware of any hardware lock.

Device S/N is 365931653133.
Is there a way to validate thus S/N is a genuine device and then fix the issue?

Thanks!
pit

samuelsadok, 2024-12-03:
The official Web GUI does not support ODrive v3.6, even if genuine. However `odrivetool` continues to support ODrive v3.6 and might also work with the replaced MCU.

The warning has to do with hardware version information that is stored on the MCU, that would be missing if you replace the MCU. I don't remember if this causes any problems from the top of my head but you can try if you can spin a motor without it.

0xPIT, 2024-12-03:
oh, thanks a lot for clarifying!

closed 2024-12-03 by 0xPIT